GivEnergy/giv_tcp #21: Unable to get giv_tcp running in Portainer with GivEnergy  GIV-AC3.0

Jim-Battle67, 2022-12-14:
Novice at Git, Docker, Compose etc so likely user error but....

Using Portainer to deploy package.

Logs just repeat the following errors and am at a loss at what I am not specifying correctly in the ENV variables. I have tried leaving the invertor IP out, putting it in. Same with MQTT params and even putting in bogus values to see if I can get more specific information. The logs really dont tell me what it is failing to connect to - I am guessing the Invertor but not sure.

2022-12-10 16:48:35,712 - startup - [CRITICAL] - No config directory exists, so creating it...
2022-12-10 16:48:35,713 - startup - [CRITICAL] - Running Redis
2022-12-10 16:48:35,719 - startup - [CRITICAL] - Running RQ Dashboard on port 9181
2022-12-10 16:48:35,719 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2022-12-10 16:48:35,755 - startup - [CRITICAL] - Recreating settings.py for invertor 1
2022-12-10 16:48:35,756 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2022-12-10 16:48:35,757 - startup - [CRITICAL] - Running Invertor read loop every 10s
2022-12-10 16:48:35,758 - startup - [CRITICAL] - Subscribing Mosquitto on port 1883
2022-12-10 16:48:35,759 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2022-12-10 16:48:35,809 - startup - [CRITICAL] - Setting daily charge target forecast job to run at: 00:20
[2022-12-10 16:48:36 +0000] [15] [INFO] Starting gunicorn 20.1.0
[2022-12-10 16:48:36 +0000] [15] [INFO] Listening at: http://0.0.0.0:6345 (15)
[2022-12-10 16:48:36 +0000] [15] [INFO] Using worker: sync
[2022-12-10 16:48:36 +0000] [18] [INFO] Booting worker with pid: 18
[2022-12-10 16:48:36 +0000] [19] [INFO] Booting worker with pid: 19
[2022-12-10 16:48:36 +0000] [20] [INFO] Booting worker with pid: 20
RQ Dashboard version 0.6.0
 * Running on 0.0.0.0:9181
 * Serving Flask app 'rq_dashboard.cli'
 * Debug mode: off
2022-12-10 16:48:37,474 - read - [INFO] - ----------------------------Starting----------------------------
2022-12-10 16:48:37,476 - read - [INFO] - Getting All Registers
2022-12-10 16:48:37,477 - read - [INFO] -  setting lock file
2022-12-10 16:48:37,478 - read - [INFO] - Connecting to: 
2022-12-10 16:48:37,617 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
2022-12-10 16:48:38,119 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
2022-12-10 16:48:38,620 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
2022-12-10 16:48:39,122 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
2022-12-10 16:48:39,624 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
2022-12-10 16:48:40,126 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
2022-12-10 16:48:40,628 - GivLUT - [CRITICAL] - Consecutive failure count= 1
2022-12-10 16:48:40,629 - read - [ERROR] - Error collecting registers: (<class 'KeyError'>, KeyError(HR:013), <traceback object at 0xb5833f08>)
2022-12-10 16:48:40,629 - read - [INFO] - Removing lock file due to read erro

Stack File is as follows.

version: "3.9"
services:
  GivTCP:
    image: britkat/giv_tcp-ma:latest
    network_mode: "host"
    environment:
      - INVERTOR_IP=192.168.86.83     # Set this to the IP address of your Invertor on you rlocal network
      - NUM_BATTERIES=1               # Number of battery modules installed and connected to the above invertor
      - MQTT_OUTPUT=True              # "True" if you want to publish your data to MQTT, "False" otherwise
      - MQTT_ADDRESS=                 # IP address of an existing MQTT broker, or leave as "127.0.0.1" to use the internal broker
      - MQTT_USERNAME=               # Username of your existing broker, if needed. Not required for internal broker
      - MQTT_PASSWORD=               # Password of your existing broker, if needed. Not required for internal broker
      - MQTT_TOPIC=                   # Root topic to publish data to. If left blank it will default to GivEnergy/<invertor_serial_number>
      - MQTT_PORT=1883                # Port of your existing broker, leave as "1883" for internal broker
      - LOG_LEVEL=Debug               # Level of logs to be reported: "Error", "Info" or "Debug"
      - DEBUG_FILE_LOCATION=          # Location of log file stored inside the container, defalt location is /app/GivTCP
      - PRINT_RAW=True                # If True this will publish all inverotr data unprocessed as well as standard data
      - SELF_RUN=True                 # If True the container will self-run and connect and publish data. If "False" the you will need to trigger externally via REST
      - SELF_RUN_LOOP_TIMER=10        # Wait time between every read command to the invertor
      - INFLUX_OUTPUT=False           # "True" if you want to publish your data to InfluxDB, "False" otherwise
      - INFLUX_URL=                   # URL of an external Influx instance
      - INFLUX_TOKEN=                 # Access Token for your Influx instance
      - INFLUX_BUCKET=                # Data Bucket of your Influx instance you want data sent to
      - INFLUX_ORG=                   # Influx instance Organisation
      - HA_AUTO_D=True                # If True (and if MQTT_OUTPUT is True) this will publish Home Assistant Auto Discovery messages to the broker
      - GIVTCPINSTANCE=1
      - PYTHONPATH="/app"
    restart: always
    privileged: true

logicnerd, 2022-12-31:
Ditto - I have the same problem as a result of trying trying to deploy GivTCP in a docker container. I tried multiple approaches to launching GivTCP in a container, including using docker from the command line. All result in the same problem as above with the same error i.e.  _"sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve".
 
Looking elsewhere for similar errors, it appears that there can be some issues with DNS name resolution inside docker containers. A workaround for this is apparently to use the docker flag --add-host=IP_Address. However, the debug information for GivTCP does not provide enough information to determine what address it is having problems resolving.  I am wondering if the IP address is actually being passed to GivTCP internally given that the debug information says "Connecting to:" without saying what it is connecting to. Does  this mean that it has a null address i.e. the ENV variable for the Inverter IP address has not been set for some reason?

G6EJD, 2022-12-31:
What IP address did you enter in the GivTCP configuration.

logicnerd, 2022-12-31:
The one for the inverter. I know its correct as you can actually log into the inverter with that IP address and change some of the settings on the Wifi dongle.

G6EJD, 2022-12-31:
Also have you tried to ping the inverter dongle at its IP address from the host platform and then with the Port number, although the latter may not respond.

Jim-Battle67, 2022-12-31:
I have tried with the inverter IP and leaving it blank to see it would be discovered. I have tried giving a name rather than IP. I have tried giving the wrong IP address. Exact same message each time.

I am not sure the message relates to the inverter or something else - I am guessing it does but not sure.

logicnerd, 2022-12-31:
> 

I can ping the Inverter IP address from the docker console without a port number - and it responds fine. I am not sure what the port number is for the Inverter i.e. what port GivTCP uses to access the data.

logicnerd, 2022-12-31:
Ref "and leaving it blank to see it would be discovered"....I don't believe this will work. Apparently the container network interface needs to be in "host" mode otherwise GivTCP will not be able to discover the Inverter IP address. By default it is in bridge mode. You can't change it to host mode. If it is in bridge mode then you have to set the ENV variable for the Inverter IP Address. At least that is my understanding; please correct me if I am wrong.

Jim-Battle67, 2022-12-31:
> 

The container is in Host mode and I enabled privileged access. This was needed for Home Assistant to do discovery , so figured I would add it.

logicnerd, 2022-12-31:
Success (at least for this current step). I found a different docker compose file, edited that to match my configuration, and I no longer have the previous error. I can also now use MQTT Explorer and see my data. This would benefit from a little investigation to figure out what the issue is and to add this to the documentation so that others don't run into the same issue. For reference, the compose file I used as the basis of a working version came from here: https://wiki.terravolt.co.uk/docs/givtcp-installation-version-2-0/

logicnerd, 2022-12-31:
Hope this potentially helps with debugging this issue. Here is the comparison between the log files for the failed attempt and successful attempt:

Failure log file entry

2022-12-31 15:20:50,062 - read - [INFO] - ----------------------------Starting----------------------------
2022-12-31 15:20:50,074 - read - [INFO] - Getting All Registers
2022-12-31 15:20:50,075 - read - [INFO] -  setting lock file
2022-12-31 15:20:50,075 - read - [INFO] - Connecting to: 
2022-12-31 15:20:50,147 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve

Success log file entry

2022-12-31 18:57:00,681 - read - [INFO] - ----------------------------Starting----------------------------
2022-12-31 18:57:00,681 - read - [INFO] - Getting All Registers
2022-12-31 18:57:00,681 - read - [INFO] -  setting lock file
2022-12-31 18:57:00,681 - read - [INFO] - Connecting to: 192.168.1.XXX
2022-12-31 18:57:00,721 - sync - [DEBUG] - Connection to Modbus server established. Socket ('172.20.0.2', 57081)

Note: XXX - deleted the final part of my Inverter IP address.

logicnerd, 2022-12-31:
Don't believe this. Could it be this simple (which would correlate with my initial suspicion that the IP address was not being passed)?

I noticed that the original [docker-compose.yml](https://github.com/GivEnergy/giv_tcp/blob/main/docker-compose.yml) has the ENV variable defined as follows: INVERTOR_IP=192.168.2.146. 

In the alternative I used with success - it has the ENV variables of INVERTOR_IP_1 and INVERTOR_IP_2. 

I therefore edited my working configuration to change it from INVERTOR_IP_1 to INVERTOR_IP - and this does indeed cause it to fail with the previous error.

Does the github documentation need to change to reflect this apparent change of ENV variable naming?

Jim-Battle67, 2023-01-01:
That did the trick ! Simply changed INVERTOR_IP to INVERTOR_IP_1 and I now see everything in Home Assistant.

I never managed to see that Wiki in all my Googling, so that is great...but there are a lot more options mentioned in the 'unofficial' Wiki than in the official git repo :-)

Anyway, minor edit and everything fired up so simply..thanks for tracking this down.

logicnerd, 2023-01-01:
Discovered that there was a non-obvious warning about this issue on the home assistant community forum i.e. "Be warned there is a breaking change for the entity and device names in Home Assistant, sorry about that but it’s there to help those with multiple invertors and batteries!"

https://community.home-assistant.io/t/givtcp-new-release-1-1-4/449070

logicnerd, 2023-01-01:
I believe this was all caused by us looking at what is an out of date version - so for others who may fall into this same trap - look here instead: https://github.com/britkat1980/giv_tcp/tree/8c15d8084eafcfaa113f00a37f5cf3e249fd90f1

Jim-Battle67, 2023-01-12:
System working great !

closed 2023-01-12 by Jim-Battle67