GivEnergy/giv_tcp #22: Only 1 battery discovered

holdestmade, 2022-12-21:
I have a AC3.0 inverter and 2x 5.2 batteries. This shows correctly in the GE dashboard.

giv_tcp only discovers 1 of the batteries

G6EJD, 2022-12-21:
In the Configuration did you select 2 batteries?

holdestmade, 2022-12-21:
This is in my logs, seems one of the serial number is just spaces 

2022-12-21 18:50:26,623 - read - [INFO] - Building battery output: 
2022-12-21 18:50:26,623 - read - [INFO] - Battery BJxxxxxxxxxx added
2022-12-21 18:50:26,624 - read - [INFO] - Building battery output: 
2022-12-21 18:50:26,624 - read - [INFO] - Battery                     added
2022-12-21 18:50:26,629 - read - [INFO] - Running the data cleansing process

holdestmade, 2022-12-22:
Just checked with the GE API and both batteries appear correctly:

```
"connections": {
"batteries": [
  {
"serial": "BJ1111111112",
"firmware_version": "3009",
"capacity": {
"full": 99.13,
"design": 102
},
"cell_count": 16,
"has_usb": true
},
  {
"serial": "BJ111111111",
"firmware_version": "3009",
"capacity": {
"full": 96.59,
"design": 102
},
"cell_count": 16,
"has_usb": true
}
```

closed 2024-10-25 by holdestmade