GivEnergy/giv_tcp #25: Gen 2 inverters

prototypeuk, 2023-02-25:
Will this work with the Gen 2 Hybrid inverters? I have tried but keep getting these errors in the logs.

- read - [ERROR] - Consecutive failure count= 11 -- (<class 'Exception'>, Exception('Garbage or failed Invertor Response'), <traceback object at 0x7fa6a70040>)
2023-02-25 12:19:44,757 - read - [ERROR] - 10 failed invertor reads in a row so removing regCache to force update...

yavinmaster, 2023-02-25:
Mine should have the installation completed on Friday.  I hope it can be
made to work.

yavinmaster, 2023-02-27:
@prototypeuk have you tried the latest version at https://github.com/britkat1980/giv_tcp?
I have this version loaded and ready for my installation once it's commissioned at the end of this week, but it would be good to know ahead of time if I'm going to run into problems.

yavinmaster, 2023-03-03:
@prototypeuk I can confirm the latest version at https://github.com/britkat1980/giv_tcp works with my Gen2 Hybrid inverter with two batteries connected.  Not done anything except observe the data so far as it is doing its battery calibration.