GivEnergy/giv_tcp #26: Mismatch between Inverter SOC and Battery SOC

Jim-Battle67, 2023-02-26:
I am noticing a bit of a mismatch between what is reported as the Inverters SOC and the Battery's own metric. It has been usual for this to differ by a % or so, but today, I am seeing the Battery showing 78% charged and the inverter showing 100% charged.

Anyone know if this normal, an issue with the integration or something I should be worried about.

It is a little frustrating if it is really only 78% charged to see KW of electricity being dumped to the grid for nothing.

G6EJD, 2023-02-26:
What does the App report when you tap on the Battery icon, then more information ? Mine says Primary 74%, Secondary 46% and total 56% there is no obvious relationship that I can see. The integration reports exactly the same values.

Jim-Battle67, 2023-02-27:
I only have one battery.

Checked the App and looks like it is consistently out of synch - so no bug in the integration software :-)

Did a reboot of the inverter and the 2 values syched up (with 1%), so appears to be a feature/bug of the battery/inverter combination.

closed 2023-02-27 by Jim-Battle67