GivEnergy/giv_tcp #27: [ERROR] - Connection to (, 8899) failed

digitlength, 2023-03-07:
Invertor wifi dongle is located at 192.168.1.95, confirmed by going to that IP address in a browser and seeing the settings.

Docker compose file:

```yaml
version: "3.9"
services:
  GivTCP:
    image: britkat/giv_tcp-ma:latest
    ports:
      - "1883:1883"
      - "6345:6345"
    environment:
      - INVERTOR_IP=192.168.1.95
      - MQTT_ADDRESS=192.168.1.212
      - HA_AUTO_D=False
    restart: always
    privileged: true
```

When starting the container with `docker compose up` I get the following:

```
[+] Running 1/0
 ⠿ Container giv_tcp-GivTCP-1  Created                                                                                                             0.0s
Attaching to giv_tcp-GivTCP-1
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,143 - startup - [CRITICAL] - Config directory already exists
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,143 - startup - [CRITICAL] - Running Redis
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,144 - startup - [CRITICAL] - Running RQ Dashboard on port 9181
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,144 - startup - [CRITICAL] - Setting up invertor: 1 of 1
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,144 - startup - [CRITICAL] - Recreating settings.py for invertor 1
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,145 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,145 - startup - [CRITICAL] - Running Invertor read loop every 5s
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,146 - startup - [CRITICAL] - Subscribing Mosquitto on port 1883
giv_tcp-GivTCP-1  | 2023-03-07 21:17:31,150 - startup - [CRITICAL] - Starting Gunicorn on port 6345
giv_tcp-GivTCP-1  | [2023-03-07 21:17:31 +0000] [15] [INFO] Starting gunicorn 20.1.0
giv_tcp-GivTCP-1  | [2023-03-07 21:17:31 +0000] [15] [INFO] Listening at: http://0.0.0.0:6345 (15)
giv_tcp-GivTCP-1  | [2023-03-07 21:17:31 +0000] [15] [INFO] Using worker: sync
giv_tcp-GivTCP-1  | [2023-03-07 21:17:31 +0000] [18] [INFO] Booting worker with pid: 18
giv_tcp-GivTCP-1  | [2023-03-07 21:17:31 +0000] [19] [INFO] Booting worker with pid: 19
giv_tcp-GivTCP-1  | [2023-03-07 21:17:31 +0000] [20] [INFO] Booting worker with pid: 20
giv_tcp-GivTCP-1  | RQ Dashboard version 0.6.0
giv_tcp-GivTCP-1  |  * Running on 0.0.0.0:9181
giv_tcp-GivTCP-1  |  * Serving Flask app 'rq_dashboard.cli'
giv_tcp-GivTCP-1  |  * Debug mode: off
giv_tcp-GivTCP-1  | 2023-03-07 21:17:32,368 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:32,870 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:33,371 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:33,872 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:34,373 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:34,874 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:35,576 - read - [ERROR] - Consecutive failure count= 5 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x7f94f2087c80>)"), <traceback object at 0x7fd131590840>)
giv_tcp-GivTCP-1  | 2023-03-07 21:17:35,791 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:36,293 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:36,795 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:37,297 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
giv_tcp-GivTCP-1  | 2023-03-07 21:17:37,799 - sync - [ERROR] - Connection to (, 8899) failed: [Errno -2] Name does not resolve
```

...and so on. I would expect to see the inverter IP address here, before the port 8899?

I have confirmed that I can ping the inverter from the device running this docker container.

I can also use `givenergy-modbus` from the Python command line from the same device to successfully retrieve data from the inverter.

Am I missing something?

yavinmaster, 2023-03-07:
It looks like you are using the docker file from britkat/giv_tcp but the older docker-composer format from GivEnergy.
Take a look at [https://github.com/britkat1980/giv_tcp/blob/main/docker-compose.yml](url).
You will see for example that the inverter ip variable is now INVERTOR_IP_1.
I've just had my system installed and got this working straight away.

digitlength, 2023-03-07:
> It looks like you are using the docker file from britkat/giv_tcp but the older docker-composer format from GivEnergy.

Yeah, I'm using the one from the repo that this issue is posted to: https://github.com/GivEnergy/giv_tcp

I didn't clock that there was another repository - thanks, I've moved over and it works!

closed 2023-03-07 by digitlength