GivEnergy/giv_tcp #28: Enable Batter Charging and Charge Level from Home Assistant

Jim-Battle67, 2023-03-10:
Created a Home Assistant dashboard with the 'Enable Charge Schedule'  and 'Target SOC' entities.

When I update these values using the GivEnergy Cloud or the Android App connected in 'Cloud Mode' they are reflected correctly in the dashboard.

If i update them in the Home Assistant dashboard, they are not reflected when I look in the GivEnergy Cloud/App. I tested to see if this is a timing issue and looked at Inverter data using the 'Remote Control' option in the browser.

I am unable to test the app connected locally as I am not present in the location.

Attached logs just in case but I dont see errors.

[_givstack-GivTCP-1_logs (9).txt](https://github.com/GivEnergy/giv_tcp/files/10945089/_givstack-GivTCP-1_logs.9.txt)

Jim-Battle67, 2023-03-10:
I did see a pull request from @lockedj  that has added a new pause between reads from the inverter. I wonder if this is part of the issue since  I do get errors when trying to set the values from the cloud - and have to resubmit.

yavinmaster, 2023-03-10:
In the version at https://github.com/britkat1980/giv_tcp there is a timer loop function which sets the time delay between inverter reads to reduce potential clashes.  Having said that, I still see the same behaviour as mentioned above but with one main difference which I don't know if you have tested - if you go in to the remote control mode and forcibly read the values you mention then they correctly read the values set from Home Assistant and then the GUI displays them correctly.  I'm guessing it is more a fault with the GivEnergy Apps and web interface itself - I'm guessing it might cache the settings and only look for chances it knows about.

Jim-Battle67, 2023-03-10:
Yeah - I tried the 'Read Category' option in the UI and still didnt get the change.

I also noticed there are no mentions of writes in the logs and just reads. Not sure if there is supposed to be some debug message on this.

Also, when I toggle the charge enable switch on, it immediately switches itself off again and then turns back on a few seconds later. Not sure if that is relevant to this problem.

yavinmaster, 2023-03-10:
As mentioned, I'm using https://github.com/britkat1980/giv_tcp and have the loop timer set to 10 seconds.  Even with this, I sometimes get a write error from the GivEnergy app, but if I submit it again immediately it usually works.  I'm guessing this is a clash between HomeAssistant and the GivEnergy app trying to access the inverter at the same time.  This is why I have the timer set to 10 seconds as a compromise between up to date data but not completely hogging the inverter comms.

Jim-Battle67, 2023-03-11:
I did a better test and it appears that setting the inverter values via Home Assistant does actually work i.e. it charges when it is supposed to.

So it appears the cloud never gets the updated values from the inverter. They must of designed it so the only expected way to set them is via the cloud API. So far have not discovered any trick to make the updated values appear in the GivEnergy cloud dashboard.

I am a little concerned that at some point the cloud will overwrite the  values in the inverter. I will monitor to see if this ever happens.

lawrencegripper, 2023-03-16:
Hit this same problem while testing things out for me. Write appears to work from the client but then never updated in the cloud app or the app. 

@Jim-Battle67 did you way the new setting was respected just not show? I'm going to try out overnight and see if works.