GivEnergy/giv_tcp #47: Not updating information from Gen1 Hybrid

Mikebert4, 2024-05-12:
Hi - amazing addon, thank you!

I've ben using this for months without issue, but a couple of days ago I stopped getting updates from my inverter.

Switching "Is old firmware" to off doesn't resolve (If set to "off" get a python error "property 'serial_number' doesn't exist".  
Switching "Is old firmware" on doesn't resolve, but it looks like it's connecting to the inverter and then... not updating?

Have restarted the inverter, and "reset to defaults" on the GivEnergy portal... no dice.

I don't have a HA backup with GivTCP below 4.3.2 to restore to, so can't tell if it's update related or not...

Debug logs: 

`2024-05-12 22:46:45,478 - Inv1 - read        -  [DEBUG   ] - inverter connection successful, registers retrieved
2024-05-12 22:46:45,479 - Inv1 - read        -  [DEBUG   ] - Beginning parsing of Inverter data
2024-05-12 22:46:45,479 - Inv1 - read        -  [DEBUG   ] - Getting Total Energy Data
2024-05-12 22:46:45,480 - Inv1 - read        -  [DEBUG   ] - Getting Today Energy Data
2024-05-12 22:46:45,481 - Inv1 - read        -  [DEBUG   ] - Getting PV Power
2024-05-12 22:46:45,481 - Inv1 - read        -  [DEBUG   ] - Getting Grid Power
2024-05-12 22:46:45,482 - Inv1 - read        -  [DEBUG   ] - Getting EPS Power
2024-05-12 22:46:45,482 - Inv1 - read        -  [DEBUG   ] - Getting PInv Power
2024-05-12 22:46:45,483 - Inv1 - read        -  [DEBUG   ] - Getting Load Power
2024-05-12 22:46:45,483 - Inv1 - read        -  [DEBUG   ] - Getting Self Consumption Power
2024-05-12 22:46:45,484 - Inv1 - read        -  [DEBUG   ] - Getting Solar to H/B/G Power Flows
2024-05-12 22:46:45,484 - Inv1 - read        -  [DEBUG   ] - Getting Grid to Battery/House Power Flow
2024-05-12 22:46:45,484 - Inv1 - read        -  [DEBUG   ] - Getting SOC
2024-05-12 22:46:45,485 - Inv1 - read        -  [DEBUG   ] - Getting Battery Energy Data
2024-05-12 22:46:45,485 - Inv1 - read        -  [DEBUG   ] - Getting mode control figures
2024-05-12 22:46:45,486 - Inv1 - read        -  [DEBUG   ] - Calculating Mode...
2024-05-12 22:46:45,486 - Inv1 - read        -  [DEBUG   ] - Mode is: Eco
2024-05-12 22:46:45,486 - Inv1 - read        -  [DEBUG   ] - New control modes don't exist for this model
2024-05-12 22:46:45,487 - Inv1 - read        -  [DEBUG   ] - Force Export is not Running
2024-05-12 22:46:45,487 - Inv1 - read        -  [DEBUG   ] - Getting Solar to H/B/G Power Flows
2024-05-12 22:46:45,488 - Inv1 - read        -  [DEBUG   ] - Getting Battery to House Power Flow
2024-05-12 22:46:45,488 - Inv1 - read        -  [DEBUG   ] - Getting Grid to Battery/House Power Flow
2024-05-12 22:46:45,488 - Inv1 - read        -  [DEBUG   ] - Getting Battery to Grid Power Flow
2024-05-12 22:46:45,489 - Inv1 - read        -  [DEBUG   ] - Getting TimeSlot data
2024-05-12 22:46:45,489 - Inv1 - read        -  [DEBUG   ] - Battery Pause timeslots don't exist for this model
2024-05-12 22:46:45,490 - Inv1 - read        -  [DEBUG   ] - Getting inverter Details
2024-05-12 22:46:45,490 - Inv1 - read        -  [DEBUG   ] - Getting Battery Details
2024-05-12 22:46:45,491 - Inv1 - read        -  [DEBUG   ] - Building battery output: 
2024-05-12 22:46:45,491 - Inv1 - read        -  [DEBUG   ] - Battery BJ2237G408 added
2024-05-12 22:46:45,493 - Inv1 - read        -  [DEBUG   ] - Night Start= Sun May 12 00:30:00 2024
2024-05-12 22:46:45,493 - Inv1 - read        -  [DEBUG   ] - Day Start= Sun May 12 04:30:00 2024
2024-05-12 22:46:45,494 - Inv1 - read        -  [DEBUG   ] - Setting Rate to Day
2024-05-12 22:46:45,497 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache`

[image]

G6EJD, 2024-05-12:
Have you tried a HA restart?

Mikebert4, 2024-05-12:
> Have you tried a HA restart?

The full works, I'm afraid: Restart, Reboot, "turn it off and go touch grass for half an hour".