GivEnergy/giv_tcp #52: Error after updating. KeyError: 'Model_1'

gazm2k5, 2024-08-18:
My instance of watchtower automatically updated some containers overnight, and now GivTCP won't find my inverter.

```
2024-08-18T09:45:51.181994218Z 2024-08-18 10:45:51,181 - startup - [INFO] - ========================== STARTING GivTCP================================

2024-08-18T09:45:53.183548847Z 2024-08-18 10:45:53,183 - startup - [INFO] - Searching for Inverters

2024-08-18T09:45:53.183611228Z 2024-08-18 10:45:53,183 - startup - [INFO] - Scanning network for GivEnergy Devices...

2024-08-18T09:46:19.850718273Z 2024-08-18 10:46:19,850 - startup - [INFO] - No inverters found...

2024-08-18T09:46:19.859795964Z 2024-08-18 10:46:19,859 - startup - [INFO] - Setting up invertor: 1

2024-08-18T09:46:19.875438945Z Traceback (most recent call last):

2024-08-18T09:46:19.875484450Z   File "/app/startup.py", line 556, in <module>

2024-08-18T09:46:19.875861437Z     createsettingsjson(inv)

2024-08-18T09:46:19.875874394Z   File "/app/startup.py", line 174, in createsettingsjson

2024-08-18T09:46:19.876016222Z     outp.write("    inverter_type=\""+str(setts["Model_"+str(inv)])+"\"\n")

2024-08-18T09:46:19.876065959Z                                           ~~~~~^^^^^^^^^^^^^^^^^^^

2024-08-18T09:46:19.876077198Z KeyError: 'Model_1'

2024-08-18T09:46:21.933319804Z 2024-08-18 10:46:21,933 - startup - [INFO] - ========================== STARTING GivTCP================================

2024-08-18T09:46:23.935024631Z 2024-08-18 10:46:23,934 - startup - [INFO] - Searching for Inverters

2024-08-18T09:46:23.935111677Z 2024-08-18 10:46:23,934 - startup - [INFO] - Scanning network for GivEnergy Devices...

2024-08-18T09:46:51.258789828Z 2024-08-18 10:46:51,258 - startup - [INFO] - No inverters found...

2024-08-18T09:46:51.265682420Z 2024-08-18 10:46:51,265 - startup - [INFO] - Setting up invertor: 1
```

G6EJD, 2024-08-18:
What setting did you make for inverter type ?

G6EJD, 2024-08-18:
Did you use the givtcp configurator? check your log file

gazm2k5, 2024-08-19:
> What setting did you make for inverter type ?

All my settings are above. I've deployed as a container, and have not used a Home Assistant addon as I don't run HA OS. I don't know what the configurator is, I'm assuming that's the front end configuration page you get in Home Assistant?