GivEnergy/giv_tcp #58: Cannot get it to work with my new 3-Phase inverter

lancer73, 2025-03-18:
Tried to get this addon working on Home Assistant. Instelled it via the repository and entered IP number and MQTT account in the config and started the addon. From the logs I gather that it isn't working. Is there something else to try?

`2025-03-18 21:46:08,215 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2025-03-18 21:46:08,217 - startup - [INFO] - Supervisor Timezone: Europe/Amsterdam
2025-03-18 21:46:10,222 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2025-03-18 21:46:15,071 - startup - [CRITICAL] - No inverters found...
2025-03-18 21:46:21,002 - startup - [ERROR] - Gathering inverter details for 192.168.107.9 failed.
2025-03-18 21:46:21,002 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:23,012 - startup - [ERROR] - Gathering inverter details for 192.168.107.9 failed.
2025-03-18 21:46:23,012 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:25,017 - startup - [ERROR] - Gathering inverter details for 192.168.107.11 failed.
2025-03-18 21:46:25,017 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:27,021 - startup - [ERROR] - Gathering inverter details for 192.168.107.11 failed.
2025-03-18 21:46:27,021 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:27,169 - startup - [ERROR] - Gathering inverter details for 192.168.107.108 failed.
2025-03-18 21:46:27,169 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:27,329 - startup - [CRITICAL] - Inverter TCXXXXXXXX which is a Gen 1 - Hybrid has been found at: 192.168.107.108
2025-03-18 21:46:31,759 - startup - [ERROR] - Gathering inverter details for 192.168.107.9 failed.
2025-03-18 21:46:31,759 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:33,764 - startup - [ERROR] - Gathering inverter details for 192.168.107.9 failed.
2025-03-18 21:46:33,764 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:35,768 - startup - [ERROR] - Gathering inverter details for 192.168.107.11 failed.
2025-03-18 21:46:35,768 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:37,772 - startup - [ERROR] - Gathering inverter details for 192.168.107.11 failed.
2025-03-18 21:46:37,772 - startup - [ERROR] - Unable to interrogate inverter to get base details
2025-03-18 21:46:37,920 - startup - [CRITICAL] - Inverter TCXXXXXXXX which is a Gen 1 - Hybrid has been found at: 192.168.107.108
2025-03-18 21:46:37,921 - startup - [INFO] - Searching for Inverters again
2025-03-18 21:46:37,922 - startup - [CRITICAL] - Running Redis
2025-03-18 21:46:37,924 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2025/03/18 21:46:37 [notice] 189#189: using the "epoll" event method
2025/03/18 21:46:37 [notice] 189#189: nginx/1.20.2
2025/03/18 21:46:37 [notice] 189#189: OS: Linux 6.12.18-haos
2025/03/18 21:46:37 [notice] 189#189: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2025-03-18 21:46:37,943 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2025-03-18 21:46:37,943 - startup - [CRITICAL] - Running Invertor (192.168.107.108) read loop every 15s
2025-03-18 21:46:37,943 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2025-03-18 21:46:37,944 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2025/03/18 21:46:37 [notice] 189#189: start worker processes
2025/03/18 21:46:37 [notice] 189#189: start worker process 197
2025/03/18 21:46:37 [notice] 189#189: start worker process 198
2025/03/18 21:46:37 [notice] 189#189: start worker process 199
2025/03/18 21:46:37 [notice] 189#189: start worker process 200
[2025-03-18 21:46:38 +0100] [196] [INFO] Starting gunicorn 23.0.0
[2025-03-18 21:46:38 +0100] [196] [INFO] Listening at: http://0.0.0.0:6345 (196)
[2025-03-18 21:46:38 +0100] [196] [INFO] Using worker: sync
[2025-03-18 21:46:38 +0100] [201] [INFO] Booting worker with pid: 201
[2025-03-18 21:46:38 +0100] [202] [INFO] Booting worker with pid: 202
[2025-03-18 21:46:38 +0100] [203] [INFO] Booting worker with pid: 203
2025-03-18 21:46:38,407 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2025-03-18 21:48:18,527 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
 Live`

More details:
Inverter model: GIV-3HY-11
Software version: DA0.007-DD0.007-A0.007

lancer73, 2025-03-21:
Beta version contains the solution.

closed 2025-03-21 by lancer73

ddowding, 2025-08-10:
Ah I see, is that on a particular branch or tag?